rotorflight/rotorflight-firmware #13: iFlight_F411_PRO no Blackbox

jasc76, 2022-04-15:
**Describe the bug**
Blackbox is not found with this FC. 

**To Reproduce**
Flash 4.2.12 20220401; Apply Defaults; BB not found
Flash Betaflight 4.3RC5 - BB chip found

**Expected behavior**

**Screenshots**
[image]

rotorflight, 2022-05-02:
Evidently the BB chip in question is not supported by BF 4.2, thus not supported by RF 4.2 either.
Will consider to add it later, either to 4.3 based RF, or as an addition to 4.2.

jasc76, 2022-05-04:
I was just diggin through some BF stuff and saw this:
https://github.com/betaflight/betaflight/pull/11537
Maybe it just needs to be activated in the board definition

jasc76, 2022-05-05:
winbond, is that the correct chip
[image]
?

jasc76, 2022-05-05:
I think its [W25Q64JVXGIM]

jasc76, 2022-05-05:
test build installed and short bench test. Logging, download, opening in BB works

rotorflight, 2022-05-05:
Fix will appear in 4.2.12-RC2

closed 2022-05-05 by rotorflight