rotorflight/rotorflight-firmware #15: Dynamic notch filter

rtlopez, 2022-07-05:
Hi,

I'm just setting up my first rotorflight helicopter and I wonder, why dynamic notch is disabled `USE_GYRO_DATA_ANALYSE` . Is it problematic in helicopter configuration or rather you plan to enable it later? i think it will do a good job in setups without rpm measurement.

https://github.com/rotorflight/rotorflight-firmware/commit/09f42713be72bee518ade30e493fc21e5838a256#diff-d33783862cc8c9583b6f155ba705b5272895c781f296766d05122b94f77a3e2fR303

Thanks, I really appreciate your work, looks promising, good job ;)

rotorflight, 2022-07-15:
Dynamic notch is disabled because it can't filter out enough harmonics, and it is using more CPU cycles than we can spare. In other words, it does not do the filtering properly, and at the same time it is harming the other functions too.
RPM notch filters do it properly without the side effects. If you have a small heli, you are probably using a D-shot capable ESC, so you have the RPM information available. With a large heli you probably want the governor, so you will have the RPM available anyway. 

In the end, we have not found a realistic use case for not having RPM info available.

If you have a realistic use case, please describe it here in the comments.

rtlopez, 2022-07-17:
Thanks for your response. I'm building 450 class heli, and it is hard to buy rpm sensor locally now. Can you tell me, if such one would work https://www.hobbywingdirect.com/products/rpm-sensor

rotorflight, 2022-07-17:
That is the one most people are using. Alternatively you can have an ESC with the "yellow" wire, i.e. the RPM output.

rtlopez, 2022-07-17:
Thanks a lot :)

closed 2022-08-05 by rotorflight