rotorflight/rotorflight-firmware #17: Add ALTITUDE_HOLD Flight Mode

rotorflight, 2022-12-29:
Add a new flight mode ALTITUDE_HOLD.

As the name suggests, the altitude is stabilised and kept constant, while the collective stick is centered.
Off-center collective would cause climb/decent, with a speed indicated by the collective stick deflection.

This flight mode can be used together with ACRO, ANGLE, HORIZON and TRAINER modes.

jabacz, 2024-01-08:
I think, the collective action is totally useless. I had this mode on film making hexacopter, never really used. It confusing for the pilot. Better was go freeflight, find position, lock. While the GPS front/back + sideways + yaw are welcome in GPS mode, the altitude change by stick tin hold mode was useless. It's almost impossible to  catch the "zero" on collective, you will always have some offset.
... Maybe having some center insensitive/hold range and change altitude only in last 20% of collective travel?
... Just experience and opinion.

panukornb, 2025-05-21:
Flywing H1 has ALTITUDE_HOLD feature.

However, I feel people who use RotorFlight are more hardcore flyers.

I am not sure if ALTITUDE_HOLD will be used by most people.

jabacz, 2025-05-22:
I'm using it regularly
Altitude hold + Angle -> time to look at telemetry without landing .... mostly capacity consumed