rotorflight/rotorflight-firmware #19: Add Vibration Analysis / Report

rotorflight, 2022-12-29:
Add functionality to measure and report vibration levels.

Calculate and report the following:
- Main rotor harmonic levels in the gyro signal (1,2,3,4)
- Tail rotor harmonic levels in the gyro signal (1,2)
- Motor vibration level
- Remaining total vibration level in the filtered gyro signal

venbs, 2025-03-18:
This feature would be incredibly valuable to our community.

nickthenorse, 2026-01-11:
Also seeking this type of feature. On my Brain2 helis, the ability to see a live overall "Vibration" value via telemetry on my transmitter is a great way to check for problems or blade tracking issues at the field.

robthomson, 2026-01-13:
Probably better done as an msp call than a sensor.

We have a long list of tasks to do - but will get to this eventually!

robthomson, 2026-01-22:
Needs resting to even know if it does / does not work.

Right now there is no real clue if the end results are usefull!