rotorflight/rotorflight-firmware #28: Add a method for adjusting leveling trim inflight

rotorflight, 2022-12-29:
There should be an easy way to adjust the levelling trim inflight.

Right now it's only available in the Configurator, which is a hit-and-miss approach.

robhaswell, 2023-02-22:
Other FBL have a level-trim flight mode. This is essentially just a levelling mode in which your RC commands adjust the trim instead directly controlling the attitude.

bob01, 2024-02-17:
Another option could be an "auto trim" model like VBar
- enable
- place the model in a stable hover out of ground effect above level ground (6ft +) on a dead calm day
- let go of the sticks, when the FC sees no large inputs it slowly determines the cyclic and yaw trim values to stop slow lateral drift and yaw
- FC interrupts sampling on large control inputs or if autorotation is engaged or motor not running
- when results are seen to be good enough flip the auto switch, land, disable 'auto trim'
- done - usually less than 5 minutes in ideal conditions (dead calm)