rotorflight/rotorflight-firmware #47: PT1 filter gain formula

rtlopez, 2023-09-14:
**Describe the bug**
I think PT1 filter gain formula is wrong
```c
float pt1FilterGain(float cutoff, float sampleRate)
{
    float omega = tan_approx(M_PIf * cutoff / sampleRate);
    float alpha = 2.0f / (1.0f / omega + 1.0f);
    return alpha;
}
```
I was courious how it works, so I've calculated some results and compared to previous version
```c
float pt1FilterGain(float f_cut, float dT)
{
    float RC = 1 / (2 * M_PIf * f_cut);
    return dT / (RC + dT);
}
```
I noted that characteristics is slightly diferrent, especialy if the cutoff is above 0.1 of sample rate. And at the 0.25 of sampling rate is equal 1 and raises.

**Expected behavior**
PT1 gain shouldn't exceed 1.0, on a picture below you can see, that it equals 1 at 0.25 of sampling frequency. On the second picture you can see that filter starts oscillating if k > 1.

**Screenshots**
[image]
[image]

**Additional context**
I don't know why PT1 filter gain formula is diferrent than in Betaflight. But if I modify it slightly to
```c
float alpha = 1.0f / (1.0f / omega + 1.0f);
```
then gain is lower than 1 for cutoffs below nyquist.

Here is the link with my calculations https://docs.google.com/spreadsheets/d/1Nv-octpMxmQeqS8wSkmaLvypIiryuTpIMJ8j50X9rwg/edit?usp=sharing

I hope this helps.

rotorflight, 2023-09-14:
The formula used in BF is incorrect. They have discussed it, and concluded that they did not want to fix it, as people are used to have it that way. Everybody would have needed to fix their configs after that.

The BF formula is for a continuous domain first order filter. When a continuous filter is converted into a discrete filter with Tustin/bilinear transform/zero order hold/etc, frequency warping happens. The cutoff formula must do frequency pre-warping to compensate.

Further, a discrete first order filter can't have cutoff above Nyquist / 2. (unlike the continuous first order).
If you try to use a freq higher than this, the gain will be incorrect.

rotorflight, 2023-09-14:
See https://en.wikipedia.org/wiki/Bilinear_transform
Section "Frequency Warping"

rtlopez, 2023-09-14:
@rotorflight thanks fo quick response, so that `limitCutoff()` is probably wrong, there is
```c
    // 95% of Nyquist
    return fminf(cutoff, 0.475f * sampleRate);
```
but should be 
```c
    // 47.5% of Nyquist
    return fminf(cutoff, 0.2375f * sampleRate);
```
I'm not sure if UI account this, it shouldn't allow to set any LPF above nyquist/2 too.

rotorflight, 2023-09-14:
Only the first order LPF has this limit. Higher order LPFs can go up to Nyquist.

rotorflight, 2023-09-14:
I'll keep this issue open until the limit has been added.

rotorflight, 2023-09-14:
Correction. Only the filters that are converted to digital domain with the "rectangular rule" (zero-order hold) rather than the "trapezoidal rule" or bi-linear transform. PTx filters are the only ones in RF like this.

Both types require the frequency pre-warping. However, PTx filters just can't go over Fs/4.

rotorflight, 2023-09-16:
Fixed in devel branch.

closed 2023-09-16 by rotorflight