rotorflight/rotorflight-firmware #74: OSD issues

LuHoang, 2024-02-05:
I was wondering did we removed the OSD features/target on the devel rotorflight 2.0 

I was following the wiki page to set up OSD, but it OSD_CS, and max7456 came out to invalid in CLI, so I'm assuming it got removed at some point.

resource OSD_CS 1 A15
feature OSD
set max7456_spi_bus = 3

https://github.com/rotorflight/rotorflight/wiki/Adding-OSD-support-for-FPV-Setups

rotorflight, 2024-02-05:
OSD support is disabled, as it does not have a maintainer with the relevant hardward.

Kingfrezz, 2024-04-16:
Willing to start some dev

rotorflight, 2024-04-20:
Feel free to enable it and start working on it.
First thing is to get it working correctly and tested, especially that it does not interfere with flight performance.
The OSD support is rather heavy, and it is mandatory that it does not cause any trouble.

The code in firmware master is from BF-4.3.x, so it should be mostly ok. except anything that can be interfering with the new RF features.

The code in the Configurator is older, and has not been maintained. It needs to be updated to the same level as the firmware.

Once this is all done and tested, then we can consider how to integrate it with the official RF releases.

Kingfrezz, 2024-04-20:
🙏 cool what was issue that it was giving

V.dennie

Kingfrezz, 2024-04-21:
Hello Rotorflight/Rotorflight-Firmware

Does Osd require built in video on board or can a mavlink external osd
connect via mavlink telemetry

V.dennie

rotorflight, 2024-04-22:
Please discuss this in the Discord. I doubt any OSD users are reading these.