rotorflight/rotorflight-firmware #91: Refactor Governor for Nitro

etocii, 2024-03-25:
1. The main distinction of nitro engine is that it requires an "idle" mode where the motor spins slowly and the clutch is not engaged. For simplicity, I believe that the "Arm" switch should be safe enough to enable the "idle" mode where a significant effort is required to start the engine (glow plug and starter) so that safety should not be a concern here.
2. The idle mode needs some special attention. Let's say that we want idle mode enabled when the FC is armed but the throttle is still on hold. There should be an easy way to trim the idle output position (either by a trim knob or some people like to use the collective stick to output throttle position directly.) It is necessary because due to the different weather, fuel, temperature, the ideal idle throttle position might be slightly different.
3. There is a problem with starting the motor as well. I found that the "start up" and "spool up" logic fails here since the rpm will have valid readout as soon as the motor is in idle. In addition, I found that at low rpm output, especially when the clutch is not engaged yet, the governor could be over-gained which results in rpm oscillation during spool up. Some ways to fix it is to engage PID loop after the output throttle or RPM has reached a minimum level to ensure the clutch is engaged.
4. There is a notion that for nitro to work well, the output throttle should not be lower than a certain value. Therefore, we should have a "min output throttle" similar to "max output throttle" we currently have.
5. (optional) Since nitro usually requires a lot of gov feedforward to work properly, it would be nice to have a way to visualize the gov FF effect on the throttle servo. One way to do it is to enable throttle FF when throttle is active and the spool up is completed even though the motor is not spinning.

Nooby1, 2024-04-08:
A fallback flight mode dependent throttle curve incase the RPM sensor gives out invalid data / RPM is outside expected range / fails is also a standard feature of an IC gov

tomaskir, 2024-04-09:
I want to add that being able to enable / disable the Gov in different flight modes / banks is super important for Nitro as well. The gov can mask effects / tendencies of badly tuned engines. For example, I always do engine tuning (carb needle tuning) with gov off, and only enable the gov after the engine has been tuned.

There is currently no way in RF to have gov on/off dynamically.

Eldr0no, 2024-06-09:
The logic Futaba have used for decades for their IC gov is very sound and would be good to look at for ideas when implementing any nitro or IC mode for RF gov. A few points to note from their system. There is a specific mode for IC engines in their gov that has certain IC specific logic and parameters.

1. Throttle curves are used in as initial throttle controls and to get rpm within range for gov to take over safely and smoothly. These also provide a safe fallback should some issue arise with gov loop and it disengages.

2. Once throttle servo endpoints and physical linkage have been set in TX for full travel without binding, the gov is calibrated by teaching it ‘Idle’ and ‘Full’ throttle positions.

3. The gov has effectively 3 possible states. A) Off B) On or ‘armed’ C) Engaged. The change of state from off to armed can be via a switch on radio or a ‘stick switch’ which is a throttle percentage above which the gov arms and below which it disarms. The gov only ‘engages’ fully and takes over RPM control once the incoming rpm is above 70% of the set target rpm. It then remains engaged until switched ‘off’ by either a dedicated channel switch or throttle percentage switch. 

4. There is a low limit throttle percentage that can be set to prevent the engine returning to an idle when head is unloaded etc - this keeps the engine ‘hot’ for returning fast to full power as needed when reloading head.

I think these are the main IC specific concepts/points that are good to keep in mind when designing an IC gov mode for RF.

bionicbone, 2024-08-30:
I propose incorporating VBAR's simplified method, which i believe could leverage existing Rotorflight code. This method uses throttle input to switch between Throttle Passthrough and Governor mode.

Throttle ≤ 50%: Throttle is passed through, allowing engine start and idle adjustment via a pot.
Throttle > 50%: Governor mode engages, ramping up to the set RPM in the current profile.
For auto-rotation practice, a switch drops the throttle back to the pot value, preventing engine cut-off and allowing controlled descent.

This approach eliminates the need for a special idle profile and supports electric setups by keeping the throttle curve at 100%. 

I haven't yet tested Rotorflight on a gas engine, so the current governor's effectiveness with gas engine lag is still uncertain for me.

pmattila, 2025-11-07:
Merged to master