rotorflight/rotorflight-firmware #103: Add RF specific tuning values to CMS

rotorflight, 2024-04-23:
Let's compose a list here.

rotorflight, 2024-04-23:
P I  D F O B gains

rotorflight, 2024-04-23:
Governor parameters:
- Full headspeed
- PID master gain
- P I D F gains

rotorflight, 2024-04-23:
Rescue parameters:
- pull-up collective
- climb collective
- hover collective

Additional:
- pull-up time
- climb time
- leveling gain
-

rotorflight, 2024-04-23:
Additional tuning parameters (if space in CMS permits):
- Cyclic cross-coupling gain
- I-term relax cutoffs
- PID controller bandwidths (nine of them)

AndersHoglund, 2024-04-24:
> Most important for tuning:
>     * P I  D F O B gains
>     * Yaw stop gains CW / CCW
>     * Yaw Collective (and cyclic) feedforward
>     * Yaw TTA gain

Done, #104 1.st commit.

AndersHoglund, 2024-04-24:
> Governor parameters:
>     * Full headspeed
>     * PID master gain
>     * P I D F gains

Done, #104 second commit

AndersHoglund, 2024-04-27:
Rescue parameters added in #107:
* pull-up collective
* pull-up time
* climb collective
* climb time
* hover collective
* leveling gain
* flip gain

AndersHoglund, 2024-04-28:
All remaining items added  #107

rotorflight, 2024-06-13:
All done for now!

closed 2024-06-13 by rotorflight