rotorflight/rotorflight-firmware #115: Add heading configuration option and magnetometer/compass telemetry

blopa1961, 2024-05-16:
The Matek M10Q-5883 connected to the G747-HELI provides both GPS and magnetometer/compass (via the QMC5883L).
The problem is that once the magnetometer is enabled (yes, it works!) it interferes with the G474's onboard gyros because Rotorflight takes the readings from both sensors and mixes them, which results in a total loss of heading.
For this reason, it is now necessary to add an option to select from where the heading reading comes from, either the G474's gyros or the QMC5883L compass, but not both.
Also, there's no option for magnetometer/compass telemetry in Rotorflight Configurator's receiver menu, only "heading" from the gyro (or maybe the mixed values). A new receiver telemetry item with FrSky's telemetry ID 0x084F could be added.

rotorflight, 2024-06-13:
The magnetometer should improve the heading precision, certainly not mess up with it.
Probably your compass configuration / orientation is incorrect, or it is not calibrated correctly.

The telemetry is another thing totally. It depends on the Tx system if it support passing compass heading.

blopa1961, 2024-06-13:
Ok. I'l try to calibrate the magnetometer.
About the telemetry, I have 2 FrSky radios; both support compass telemetry, for which the expected sensor ID is 0x084F.

There's no Compass option in Rotorflight Configurator.
[image]

Is "heading" the correct option?
iNav's compass does not work in my current config, maybe I'm doing something wrong...

blopa1961, 2025-03-27:
@plesogor what is it you assigned me?

pmattila, 2025-03-29:
There is a heading sensor already. Not sure what this add...?

robthomson, 2025-03-29:
I think this was raised inon 2.1,  the whole frsky stuff is different now with more sensors.   Maybe best to have @blopa1961 review after RC1?

blopa1961, 2025-03-29:
> There is a heading sensor already. Not sure what this add...?

The difference is that "heading" comes from the gyro, which has no idea where the north is, and "compass" comes from a magnetomenter like the QMC5883L.
I published an S.PORT compass, you might like to take a look here:
https://www.instructables.com/An-SPORT-Compass-for-FrSky-Radio-Control-Telemetry/

blopa1961, 2025-03-29:
An by the way, this is not just theoretical, I'm using this GPS/mag with both my Nexus and my G474-HELI:
https://www.mateksys.com/?portfolio=m10q-5883

ZoofPL, 2025-03-31:
> And by the way, this is not just theoretical; I'm using this GPS/mag with both my Nexus and my G474-HELI: https://www.mateksys.com/?portfolio=m10q-5883
> 
> iNav LUA'a compass doesn't work

Is there a way to connect the MQ10's compass to the Nexus as I'm using ELRS. It requires I2C......

pmattila, 2025-03-31:
The big question here is, why do anybody need the raw data from the compass?
The heli is moving so fast the compass can't generally keep up with the directions.
The FC has best of the class fusion algorithms to remedy this. As a result the heading we are using is actually correct and useful, unlike the raw compass data.

ZoofPL, 2025-03-31:
Good question, I should have explained. 
I'm  logging/recording all my flights with a speedybee running ardupilot with a M10Q GPs + compass. With a script it allows me to replay the flights almost exactly in the Heli-X simulator including stick-inputs and of course heading. 
It helps analyse and improve.
So it is a logging application.
I'm migrating to RF2 on the Nexus and it would be nice if the full logging can now simply be done by the Nexus, but then I need to also log the compass.
If this were possible it could someday become a standard feature.

blopa1961, 2025-03-31:
> Is there a way to connect the MQ10's compass to the Nexus as I'm using ELRS. It requires I2C......

Connect the 5883 to port C:
Pin1: (mapped to TX by default) = SCL, Pin2: (mapped to RX by default)= SDA, Pin3: 5V, Pin4: GND

in CLI enter:
resource SERIAL_RX 3 NONE
resource SERIAL_TX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
save

blopa1961, 2025-03-31:
> The big question here is, why do anybody need the raw data from the compass? The heli is moving so fast the compass can't generally keep up with the directions. 

[image]

As I said, I use iNav LUA to display telemetry in my radio. The compass info (26° in this photo) comes from the compass "raw data".

pmattila, 2025-03-31:
Is there a particular reason why you want the actual compass data, rather than the more precise Heading?

pmattila, 2025-03-31:
0x084F is GPS Heading (map heading) in FrSky telemetry. If you have a GPS, it will be used.
It's not the same as compass heading.

blopa1961, 2025-03-31:
> 0x084F is GPS Heading (map heading) in FrSky telemetry. If you have a GPS, it will be used. It's not the same as compass heading.

Ok, if the heading information combined with a working compass like the QMC5883L is accurate regarding the cardinal points I'll just rename the sensor to whatever iNav LUA requires

pmattila, 2025-03-31:
The Heading sensor is kept separate, because it is not Map/GPS heading, but magnetic / pseudo heading.
In case there is no compass available, it is relative to the startup direction (with slow drift). With a compass, it is compass north heading.

ZoofPL, 2025-04-27:
Using M10Q with 5883. Connected to the Nexus per instructions above (I2C) and calibrated. The sensors do provide data. However, it does not make sense (yet). The heading displayed in the configurator is off by 10s of degrees. Rotating the heli gives strange and lagging rotations on screen. The sensor output under Sensors has a lot of coupling between the three axis - unclear to me how to interpret. 
 
The MQ10's arrow is facing forward, mounted in upright position. Default MAG alignment set in Configuration. With Ardupilot the M10Q configuration worked pretty straight forward and gave great results.  Unsure of what I can do to get it going with Rotorflight and the Nexus. 

On a second note: 30 seconds to do the magnetometer calibration is short for a 600 size heli - can this be extended to al least a minute?