rotorflight/rotorflight-firmware #120: Forward Programming for Hobbywing Platinum V4 ESCs

rotorflight, 2024-06-13:
HW Plat V4 programming is not implemented.

bob01, 2024-07-28:
No promises on results but I'll give this a second look in the next little while 
(ie if someone else wants to tackle this before then by all all means)

robthomson, 2024-07-28:
I will happily do testing and porting.   Little chance of me havering the low level knowledge to do the main stuff.

rotorflight, 2024-07-29:
Thanks Bob, I can test this too.

jimmy6616, 2024-09-23:
I can also test, I have a Nexus with HW V4

ArchieRC, 2025-01-25:
I can test.   I have several heli's with the HW130 and HW120 V4

robthomson, 2025-01-25:
I have been chatting with hobbywing on this.

They are not being helpful - not wanting to share protocol docs.

Apparently the v4 protocol is a mess.  Lots of encrypted comms.

Not sure if this will ever move forward.

ArchieRC, 2025-01-25:
The V5's seem to be a nice improvement.

bob01, 2025-03-27:
Sorry guys - as mentioned above v4 is a mess - anybody else wants to tackle this go ahead - lots of examples of how to do it with the existing supported ESCs - I wont be taking this on.

robthomson, 2025-03-27:
I think we should close this.

I had a dialog with hobbywing, and even they don't want to provide documentation, saying it's a mess and not really viable to do.

Will close now.  For some cases we just have to have telemetry only.

closed 2025-03-27 by robthomson