rotorflight/rotorflight-firmware #124: Accumulated Governor Refactor Suggestions

etocii, 2024-06-24:
The current governor logic is confusing and outdated and difficult to unify across systems. I believe that using 3-position logical switches for governor is the way to go.

1. Instead of throttle position from the TX, we should have a single 3-pos (HOLD) switch controlling channel 6, meaning that low position is ~1000us, middle ~1500us, high ~2000us, corresponding to motor off, autorotation or nitro engine idle, and motor on. We should also have a dedicated profile channel defaulted on channel 7 with profile 1, 2, 3.
2. The arming procedure can be simplified to that when the FC knows the HOLD switch in in "motor off" position, arming is allowed assuming no other flags.
3. The default arming position can be simplified to for example, channel 6 <1050us

There should be only 3 governor modes. Passthrough, Electric, and Nitro. No handover% is necessary
1. Passthrough mode should have no slow start or spool up as default. There should be a new setting called "autorotation throttle percentage" that can be set according to ESC manual. This results in a simple throttle output control where for example: 
hold or disarmed = 0%
autorotation = 30%
Profile 1 = 50%
Profile 2 = 65%
Profile 3 = 85%

2. Electric governor mode should have each profile setting their own target RPM. D gain should be hidden and default to zero. We should stop using F gain as a factor on the collective and cyclic feedforward. It is redundant. Instead, we can set collective and cyclic precomp directly in the configurator. There should be a toggle to trigger whether battery voltage is used for the governor. 
For example: 
hold or disarmed = 0%
autorotation = 10%
Profile 1 = 1500rpm
Profile 2 = 1800rpm
Profile 3 = 2100rpm

There should also be an option to limit how much P and I gains are acting on the output and charged. The P should have positive and negative limit set separately, and I should have its limit. This is to reduce the effect of too much motor torque jerk which results one way disengage results in tail kick.

3. For the Nitro governor, Profile 2 and 3 should be the same. However, profile 1 should reuse the collective stick to control throttle output from the "autorotation throttle percentage" to full throttle. This is convenient when starting a motor for the first time. In addition to that, there should be a "minimum throttle" where the output throttle after spoolup cannot be below this position. Because nitro motor reacts to throttle change slow when it is running very slow, also we do not want to disengage the motor clutch during maneuvers, the minimum throttle keeps the motor running at a relatively good RPM range.

For example: 
hold or disarmed = 0%
autorotation/idle = 15% (adjustable using another channel if necessary)
Profile 1 = 15% (or whatever is set to autorotation%) to 100% depending on collective position
Profile 2 = 1800rpm
Profile 3 = 2000rpm

ahmad-kassem, 2024-06-24:
I totally agree with that. 
It is a cleaver proposal,  hope it gets implemented

AndersHoglund, 2024-06-28:
This is not refactoring, this is total redesign.
What has channel order, and hard-coded ch 6, to do with governor? Different radio systems will still have different channel orders.

etocii, 2024-06-28:
My goal is to create a unified channel mapping between Tx and Fc so that setting up and exchange models are easier. Most TX has the ability to remap channels. Governor is the problem because I think this is the last thing where people tends to have different setups. If we can unify the channel mapping for how to control governor, we could make beginner’s life way easier by defaulting on this channel mapping. Ofc you still have the option to do whatever you want but those are should be under “expert” settings.

offer-shmuely, 2024-07-20:
Interesting idea 😀
I am confused with section 1
I am using esc gov
My motor speed (low, mid, high) is not per profile 
How will it be configured?
Will it still be possible?

scooper1, 2024-07-23:
like option 1  when no rpm signal is available  - would mean I don't have to have a radio model for each none rpm signal heli

pmattila, 2025-05-24:
As a whole, this proposal would be acceptable to a minority of users only.
I shall pick up the nuggets of good ideas, and implement those in a more generic way.
Hopefully can accommodate all users and requirements.