rotorflight/rotorflight-firmware #125: Telemetry Reuse for Jeti

mikekenyon99, 2024-07-06:
Is it possible to have a similar option like the crsf_xxx_reuse for the jeti exbus so I can see headspeed on my transmitter?

Mike

mikekenyon99, 2024-08-10:
Thats a fair comment/point :-) I can see how it would be hard to test.

rotorflight, 2024-08-10:
Someone with Jeti and programming skills should look into this. Like Rob said, none of use is using Jeti.

robthomson, 2024-08-10:
Do you have a working jeti + telemetry?

It may be possible to try extend this a bit with me adding sensor - and sending you a test firmware for RF?

mikekenyon99, 2024-08-10:
Yes, currently I have a jeti tx/rx with hobbywing v4 esc.  However find the current readings very poor/inaccurate so was thinking of removing it.  However happy to try

randomgraphics, 2024-10-14:
I have Jeti DC-24 and R6L receiver that supports telemetry, with a Hobbywing Platinum 60A V4 ESC. i'd love to help testing this feature if possible. I'm a programmer too. If someone can give me a kickstart of where to begin, i can give it a try to help implementing/improving this feature.

I read the ex-bus protocol document a bit (https://www.jetimodel.com/support/telemetry-protocol/). It looks that the exbus protocol only works in half-duplex mode that requires special wiring that mix the TX and RX signal into 1 wire.

robthomson, 2024-10-14:
There is no need for a reuse function.

Jeti already can support multiple rpm sensors.

It's more that they just need to be tent.

Tero on discord is looking into this... Might be worth having a touch base with him to avoid duplication of effort?

robthomson, 2025-02-07:
Closing issue as have added additional sensors - no need for reuse

closed 2025-02-07 by robthomson