rotorflight/rotorflight-firmware #126: BB log / Auto delete, or overwrite, or rolling log so it is able to record always

azaz44, 2024-07-13:
it seems BB is now made mainly for tuning, so you can record logs to tune filters or PIDs. But recording means some manual work, at least to erase data so it has space.

However, BB would be a fantastic tool to diagnose issues, in case of heli crash or any incident when reason is unknown. Having access to recent flights at any time would be super handy. Having voltages, currents, but also servo outputs and RF actions, would help to find out very many issues. I had an umber of incidents because of servos getting stuck (because of gears). It took me a lot of time to find reasons. But this would be visible in log files, that RF is trying to push the servo all time and errors in PID accumulate.

Actually, the name BlackBox suggest such usage. Blackboxes in real planes record all the time.
As it is now, if you forget to delete your logs, then it will not log and data is lost.

rotorflight, 2024-08-12:
This would be great.
This is a large change and will need time to develop. Likely slipping into 2025.

gongtao0607, 2024-12-05:
This will be nice feature to have so you can set and forget. It will be a fair amount code change:
  1. Make the BBlog storage volume (flashfs.c) a circular volume and update all the references (MSC/emfat and the MSP log dump). (There are some tricky edge cases involved and need to update the existing flashfs framework).
  2. When arming, maintain a certain amount of free space (by erasing a few sectors). This will be usable milestone.
  3. Furthermore, we can support erase a sector in the runtime when needed. From my rough calculation, this approach should be generally fine with NAND flash, whose erase is in a few ms level. On NOR flash, it will need what is called **erase suspend** feature (which is not commonly supported) to compensate the slow erase, and we need to implement the controlling logic.

gongtao0607, 2024-12-21:
I'm working on a proof of concept "arming erase" and "background erase" logic:
https://github.com/gongtao0607/rotorflight-firmware/tree/flashfs_online_erase

On Flydragon, it works perfectly after increasing the buffer from 128 to 256 (byte) (means no buffer drop). On a Matek however, it has ~10ms buffer drop on every erase. Both have the same flash (NAND) and the actual erase time logged are always around 2ms. I'm still trying to figure out the discrepancy.

I do not have a FC that has NOR flash to test, and I don't expect it to work as of today (without erase suspend).

The arming erase should work regardless. It may locks 10ms to 1s before arming but who cares.

gongtao0607, 2024-12-26:
There's also one minor issue with the filename. I want BBLog file name to contain a datetime string, so user don't need to overwrite or rename when they keep a collection of all history logs (like me!). 

However, today's emfat library only implements DOS 8+3 filename, meaning only 8 chars in the filename (which is not enough for date + time as plain text). Long file name support seems to require a bit of work and I'm not sure if it will worth the effort.

azaz44, 2024-12-28:
> There's also one minor issue with the filename. I want BBLog file name to contain a datetime string, so user don't need to overwrite or rename when they keep a collection of all history logs (like [...]

One workaround would be to have a unique filename (might be timestamp encoded in less human readible form), but have timestamp also available as creation or modification date. Would make it easy to find flights as long as you don't mess the timestamps while copying files around.

rotorflight, 2024-12-28:
Looks like we are going for the long filename option.
Thanks gongtao0607!

gongtao0607, 2025-01-31:
This feature is implemented by: #209 #211 #224 and https://github.com/rotorflight/rotorflight-configurator/pull/251

Feel free to close

azaz44, 2025-01-31:
This is just so great, thank you for all the hard work. This is going to be a killer feature of RF. Closing issue.

closed 2025-01-31 by azaz44