rotorflight/rotorflight-firmware #173: Spektrum Avian ESC Telemetry and Forward Programming

QuPsi, 2024-09-30:


AndersHoglund, 2024-10-31:
This would require the full SRXL2 bus master protocol in RF and this is not made public by Spektrum. We only have a subset supporting incoming data from Receivers, no support for output to ESC, servos etc. 
Spektrum has published protocol support for SRXL2 bus slave only, not bus master. Specifically, `spm_srxl_master.h` is not available. Fwd programming protocol is not published either. 
https://github.com/SpektrumRC/SRXL2

QuPsi, 2024-11-01:
Yes, I spoke with some Spektrum guys... Forward Programming is definitely an internal secret sauce. That said, textgen is possible for remote programming on the ESC (this is a nice to have but really not that important).

Also, Ikon has figured out how to get telemetry data from Avian ESCs. I don't know how Ikon got telemetry working, but here's some documentation showing that they have.

From what I'm understanding from Spektrum, it needs to be plugged into a UART port. Given that the ESC is powering the servos and unit, I'm thinking plugging it into the S. BUS port would be the best option. 

https://updatebrain.mshbrain.com/software/README_for_ESC_datas_log_and_telemetry.pdf

Seems like there's documentation for how to send telemetry via SRXL2.... Need to figure out how to get the telemetry data from the ESC and then transmitted through SRXL2.

https://github.com/SpektrumRC/SpektrumDocumentation/blob/master/Telemetry/spektrumTelemetrySensors.h

robthomson, 2024-11-01:
I suspect we can get help from frankie on this.. Will link him on as he knows the fwd programing well.

frankiearzu, 2024-11-01:
This is what i have documented about the FP protocol.. have to update a few other things that i found, but has the basics:
https://github.com/frankiearzu/DSMTools/blob/main/DSM%20Forward%20Programming/Docs/DSM%20Fwd%20Prog%20Protocol.md

The challenge to use it for 3rd party flight controllers is that all the text displayed on the screen is part of the TX firmware. The protocol only send the message# to display for Menus or text values.    This is to allow localization of messages in different languages.

On Spektrum TX, is part of the firmware, so a 3rd party cannot add new messages..  You probably will have to piggyback on existing messages.

On EdgeTX, i have all the messages in a file "msg_fwrdp_en.txt" for the english language messages  (inside DSMLIB).

Thats why TextGen is probably more atractive for 3rd party, since the Text comes as part of the protocol message (8 lines, 12 chars)

AndersHoglund, 2024-11-02:
We already have SRXL2 based Telemetry output via Rx, but no tm sensor input from any SRXL2 sources. Like ESC etc.
And we already do TextGen based "Fwd Programming". CMS is using this, and TextGen can only be used by one function/device. Same as any TM sensor unfortunately, Spektrum do not support multiple sensors of the same type.
/A

AndersHoglund, 2024-11-02:
An Implementation of the real Fwd Programming protocol would be nice, as a complement to TextGen based CMS. But that would require doing the server side of FP. Current Lua implementation is client side. But I am not sure this is meaningful to do. FP only handles known (by Spektrum) entities, hard coded strings etc in the Spektrum Txs. Will be hard to map all RF specific config items to the items Spektrum knows about. But of course, we could do RF specific things in a Lua client and the firmware server using FP protocol. But that will not work with std Spektrum Tx, only with EdgeTx/Ethos etc Txs with Lua support.
/A

AndersHoglund, 2024-11-02:
The SRXL2 bus master subset we have in BF/RF was done by @MiguelFAlvarez , a Spektrum dev. Maybe ping him to do the full protocol stack for Avian ESC support. 
/A

AndersHoglund, 2025-03-27:
I think this issue/feature req. can be closed as the Avian ESCs are not really suited for RF use.

1. No fast RPM output, for filtering and governor. It would require external RPM sensor. 
2. It's text mode telemetry based config interface is rudimentary, and in conflict with CMS. The later could perhaps be resolved though, by a device selection in the opening screen. 
3. Implementing full SRXL2 Master code would be tricky without Spektrum assistance.
4. Few users.

There are other much better alternatives, offering full fwd programming and RPM signal etc etc.. Not worth the efforts needed. 
/A

QuPsi, 2025-03-27:
Only commenting to say that it does have fast rpm output. It has a separate lead for rpm. I've been using it on my machines and implemented filtering without issue.

I've spoken to Spektrum. Though they won't give us master.h, I was informed that it may be possible to reverse engineer what's needed to get full data using a logic analyzer.

Unfortunately, I just don't have the expertise to do this, otherwise I'd attempt it.

AndersHoglund, 2025-03-27:
OK, my bad. Should have checked RPM function better.
Since the Avians are based on  HW hardware, Spektrum might just have left the RPM output there. 
Not documented by Spektrum though.
/A

pmattila, 2025-11-07:
We are still happy to add the support, _IF_ we somehow get the required information.