rotorflight/rotorflight-firmware #191: Absolute control (or something similar)

egonl, 2024-11-23:
A long time ago, before Rotorflight, I was using Heliflight. It flew pretty decent, and there was one feature that I really liked: [absolute control](https://github.com/betaflight/betaflight/pull/5968):

> Abs control ensures that the attitude of the quad is brought to were it would have been without axis crosstalk ([according to the author of absolute control](https://github.com/betaflight/betaflight/issues/7796#issuecomment-474999036)).

It does so by manipulating the setpoint - a bit of a hack. But the result was nice: I could do rainbows and loopings without worrying about cross coupling. The heli just stayed in its trajectory. No (or not much) drift, and no cross coupling tuning required. Here's an [old vid](https://youtu.be/BXOB1IgYekA?t=150) where I did some rainbows. Those were very easy to do with Heliflight - and I find them harder to do with Rotorflight.
I think it would be nice to have something similar in Rotorflight.

egonl, 2024-11-23:
Comments from Dr. Rudder 2024-10-30:

Absolute Control (poor name) is an attempt to do cascaded PID loops in BF.
The inner loop would control the rate, and the outer loop would control the attitude.
This is an industry standard way of solving this kind of control problems.
In fact I have been thinking about coming back to this at some point.

About a year ago we had a discussion about cascaded loops, and back then
the opinion was "to do what the other FBLs are doing", and get that working well.
I suppose we now have that working well, so it's something to consider.

The problem with AC was the tuning. Instead of one I-gain, suddenly you have two I-like-gains,
and another "error-relax" in AC. Even BF disabled AC by default as nobody could give
a good answer how to tune it.

Now we also have HSI. I have no idea what to do with AC + HSI.
It would need a good amount of thinking and refactoring.

I am not 100%, but I think you should be able to get the same level of stability with
RF-2.1 that you used to have with HF3D with AC. 

Would you be able to take logs from both? Still have a HF3D heli around?

It would also be good to fly them back-to-back, as memory can play tricks!

egonl, 2024-11-23:
I flashed Heliflight 3D again flew with absolute control. It really makes a big difference when doing long rainbows - the heli stays more or less in the same trajectory. With Rotorflight there's a lot of drift that requires a lot of manual corrections and that I can't tune away.

Heliflight 3D [log with debug mode AC_CORRECTION](https://github.com/user-attachments/files/17882375/LOG234.-.AC_CORRECTION.TXT).

etocii, 2024-11-27:
I have some ideas about it recently as well. I think if the inner loop is good enough at detecting windspeed we should be able to utilize absolute control on the outer loop. The biggest problem that I can think about is that the outer loop P term (the other integral) will accumulate while it should be in the HSI. One solution might be to "reverse" bleed that vector into HSI as well during slow cyclic, medium to high collective, and slow collective rate condition.

gongtao0607, 2025-01-29:
https://github.com/rotorflight/rotorflight-firmware/pull/233#issue-2819558849

Please help test and report if it's useful.