rotorflight/rotorflight-firmware #230: Refactoring or Attenuation of D-Term on Roll

ArchieRC, 2025-01-25:
D gain on Roll is aggressive requiring the gain to be tuned down to 10-12.   Higher values start causing ripples affecting Pid sum for Roll.   This has been replicated over 20+ Heli's.    Pitch can be adjusted in and doesn't cause the same issues. (usually in the 45-55 Range)  A 1:1 ratio head: swash exaggerates the issue.

ArchieRC, 2025-01-25:
D term cutoff has been adjusted to see if the input for D term could be tuned.    

10Hz - Oscillations started to show there was very noticeable degradation of flight performance.  Cross Coupling was severely affected.
15Hz - Slight degradation in flight performance. 
20Hz - Best performance
25Hz - Oscillations started to show there was very noticeable degradation of flight performance.  Cross Coupling was severely affected.

This test was looking for a way to "Soften" the D term to make more active in slower movements without.  10hz and 25Hz gave very similar results.  Softening was noted.

I have BBL's that can be shared.

ArchieRC, 2025-01-25:
Slight adjustments on roll D term affects the cross coupling tuning.   Softer, more active d term could make cross coupling gain less sensitive.

ArchieRC, 2025-02-03:
@pmattila   I've shared some notes on something i've found with this.

pmattila, 2025-07-22:
@ArchieRC can you comment what is your current understanding and opinion what these values should be like?

ArchieRC, 2025-07-23:
I have been able to get very good results with tuning the bandwidths and d cut off.

When I wrote that test back in Jan I didn't test deep enough.    I have 700 flights since then tuning this and have it running on 20+ helicopters.   This is what Brian Birdsong was running on the Heli he just won Nats with.   

D Term cut off of 12 hz has worked on every helicopter i've tuned it on.   I've went down to 6 and way up high.    10 really calms down the sensitivity to oscillations.  I've found 12 to make D more active but without the oscillation side affects of the default 15.

I'm tuning all of my Heli's with Cut offs at 12, 14, 20 for RPY.    At this value they tune out between 30-38 on roll and pitch.  I'm running Boost at 35, 350 for R,P.  

The other factor that is coming into play is Bandwidth.   This is very critical for Low Head Speed.  Bandwidth for R,P works out best between the first and second harmonic.   Example 28 works well for 1000 HS.   I believe the Bandwidth should be set to default between the first and second Harmonic of the HS selected.    Or possibly to be dynamic based on HS.           (Yaw doesn't follow this pattern and seems to benefit from the higher bandwidth)

To the comments on refactoring I believe the current levels are fine once the cutoff's are tuned.    However.   I don't think it would be bad to make them 3 times weaker in the pid mode 4.     So what is 33 today would be 100.     Then also refactor Boost gain so that what is 33 today would be 100 on roll and what is 330 today would be 100.

This way you could put 100 on I, D, F, B for roll and Pitch and have a great starting point.   Default Roll D cut off to 12. and make bandwidth default based on HS.

ArchieRC, 2025-07-23:
Example tune at 1000HS

# profile 0
set pitch_p_gain = 115
set pitch_i_gain = 185
set pitch_d_gain = 45
set pitch_f_gain = 130
set pitch_b_gain = 250
set roll_p_gain = 100
set roll_i_gain = 180
set roll_d_gain = 35
set roll_f_gain = 120
set roll_b_gain = 25
set yaw_p_gain = 115
set yaw_i_gain = 280
set yaw_d_gain = 20
set yaw_b_gain = 10
set yaw_f_gain = 10
set pitch_d_cutoff = 14
set pitch_gyro_cutoff = 28
set roll_d_cutoff = 12
set roll_gyro_cutoff = 28
set yaw_cw_stop_gain = 110
set yaw_cyclic_ff_gain = 20
set yaw_inertia_precomp_gain = 20
set cyclic_cross_coupling_gain = 20
set error_decay_time_cyclic = 150
set gov_gain = 55
set gov_p_gain = 15
set gov_i_gain = 60
set gov_d_gain = 4
set gov_f_gain = 20
set gov_p_limit = 40
set gov_d_limit = 5
set gov_f_limit = 40
set gov_yaw_ff_weight = 10
set gov_cyclic_ff_weight = 35

ArchieRC, 2025-07-23:
IL Goblin running 2000      

# profile 1
set pitch_p_gain = 110
set pitch_i_gain = 115
set pitch_d_gain = 35
set pitch_f_gain = 105
set pitch_b_gain = 350
set pitch_o_gain = 60
set roll_p_gain = 95
set roll_d_gain = 35
set roll_f_gain = 95
set roll_b_gain = 35
set roll_o_gain = 60
set yaw_p_gain = 95
set yaw_i_gain = 210
set yaw_d_gain = 20
set yaw_b_gain = 20
set yaw_f_gain = 20
set pitch_d_cutoff = 12
set pitch_gyro_cutoff = 45
set roll_d_cutoff = 12
set roll_gyro_cutoff = 45
set yaw_cyclic_ff_gain = 5
set cyclic_cross_coupling_gain = 10
set error_decay_time_cyclic = 200
set iterm_relax_cutoff = 14,12,12
set gov_headspeed = 2000
set gov_gain = 50
set gov_p_gain = 18
set gov_i_gain = 60
set gov_d_gain = 5
set gov_f_gain = 14
set gov_p_limit = 40
set gov_d_limit = 5
set gov_f_limit = 40
set gov_yaw_ff_weight = 5
set gov_cyclic_ff_weight = 35

ArchieRC, 2025-07-23:
3 blade IL Goblin Pro example.   I would call this a milder tune.   The heli flys extremely smooth and quiet.   

# profile 1
set pitch_p_gain = 95
set pitch_i_gain = 110
set pitch_d_gain = 35
set pitch_f_gain = 90
set pitch_b_gain = 300
set pitch_o_gain = 60
set roll_p_gain = 80
set roll_i_gain = 105
set roll_d_gain = 35
set roll_f_gain = 85
set roll_b_gain = 30
set roll_o_gain = 60
set yaw_p_gain = 85
set yaw_i_gain = 180
set yaw_d_gain = 20
set yaw_b_gain = 5
set yaw_f_gain = 10
set pitch_d_cutoff = 12
set pitch_gyro_cutoff = 45
set roll_d_cutoff = 12
set roll_gyro_cutoff = 45
set yaw_d_cutoff = 15
set yaw_cw_stop_gain = 110
set yaw_precomp_cutoff = 6
set yaw_cyclic_ff_gain = 5
set yaw_inertia_precomp_gain = 20
set cyclic_cross_coupling_gain = 10
set error_decay_time_cyclic = 200
set iterm_relax_cutoff = 14,12,12
set gov_headspeed = 1850
set gov_gain = 55
set gov_p_gain = 20
set gov_i_gain = 55
set gov_d_gain = 5
set gov_f_gain = 12
set gov_yaw_ff_weight = 5
set gov_cyclic_ff_weight = 35