rotorflight/rotorflight-firmware #237: Telemetry for JR Propo

garfield2412, 2025-01-31:
Hi There,

some pilots use to fly JR.
[DanielGA](https://www.rcgroups.com/forums/member.php?u=771155) and myself figured out how the JR thing works - Thanks to Daniel - he is a great guy!
He made demo for an Arduino, sending one value and i modified i to a full sending version of all possible telemetry values - sending the values with half-duplex on softserial.

Maybee if the RF2 Programmers have a look here - they might implement it to RF2 somehow?

https://www.rcgroups.com/forums/showpost.php?p=53427707&postcount=151
i am no expert - chatgpt helped me - but it works fine!

garfield2412, 2025-02-26:
Sending Demo values:

https://github.com/garfield2412/jr_propo_telemetry_demo/tree/main

TheNige069, 2025-04-08:
I have been working on implementing the JR telemetry based on the work done by DanielGA and Garfield2412. Still working on how telemetry in RF is implemented.

garfield2412, 2025-04-08:
> I have been working on implementing the JR telemetry based on the work done by DanielGA and Garfield2412. Still working on how telemetry in RF is implemented.

Die you recognize the latest Updates?
Daniel had to change the way an RP2040 was sending JR Telemetry
https://www.rcgroups.com/forums/showpost.php?p=53597685&postcount=325 
(Two Stop Bits) .
Don't know If this is important for RF Hardware too 

while with our barely working arduino nano softserial code the two stop Bits where not needed. but maybee this rough code ist not sharp eniugh to see dropped telemetry values .