rotorflight/rotorflight-firmware #256: 'Throttle %' telemetry sensor is 10x expected value (FrSky telemetry only)

bob01, 2025-03-04:
Test conditions...
- branch: master
- FrSky RF007
- RF gov mode = MODE1 w/ RPM sensor disconnected

Arm and engage throttle.
Expected behavior is gov will ramp to IDLE w/ "Throttle %" settling at 20%

ELRS - "Throttle %" is 20% as expected
FrSky - "Throttle %" is 200% unexpected

[image]
[image]

robthomson, 2025-03-04:
whats the raw value being received?
in ethos sensors
it might just be that we are not adding a decimal when auto creating the sensor on ethos

bob01, 2025-03-05:
Ethos -> telemetry -  that's the first image above, no?

NightHawk32, 2025-03-05:
should be fixed with this: https://github.com/rotorflight/rotorflight-lua-ethos-suite/pull/305