rotorflight/rotorflight-firmware #280: FC freezes when selecting XDFLY

ahmad-kassem, 2025-04-05:
I think there is an issue with XDFLY telemetry option. 

I tried 2 times on 2 different controllers and still same issue. 

To replicate the issue,  start with a new FW installation,  then select esc telemetry on one of the uarts and then go to motors page and select Xdfly as telemetry protocol and save. 

After reboot when you select the motor tab the configurator displays waiting for data and then freezes there.

plesogor, 2025-04-05:
Which pin on the XDfly ESC are you using, RX or TX. There was a change for 2.2 that we have to use "RX" pin for XDfly now for bidirectional support for Fwd programming. Let's confirm if you are using RX or TX pin on the ESC.

ahmad-kassem, 2025-04-05:
> Which pin on the XDfly ESC are you using, RX or TX. There was a change for 2.2 that we have to use "RX" pin for XDfly now for bidirectional support for Fwd programming. Let's confirm if you are using RX or TX pin on the ESC.

I was using yhe TX pin on the XDFLY,  but the freeze happens with the esc disconnected also.

plesogor, 2025-04-05:
Thanks @ahmad-kassem!

This is 100% reproducible by upgrading to 2.2 RC1 changing telemetry the protocol to "XDfly" and not selecting "one-wire communication" for bidirectional. Then the FC will hang and become unresponsive. To recover put your FC into DFU mode via the boot button on top of the FC, and then reflash w/ full erase. This will be fixed after it is investigated by the team.