rotorflight/rotorflight-firmware #281: FC freezes after changing a setting while armed

egonl, 2025-04-06:
When you change a setting using Lua while armed (and probably also when making an adjustment), the FC will sometimes freeze. 

Steps to reproduce (on the S1 with G474-HLITE):
- On the Blackbox tab, set Disarm grace period to 10s
- Go to the Status tab and enable arming
- Arm the heli, make a change using Lua and save. The Lua script will give a warning: 'Settings will be saved after disarming'
- Disarm
- The FC usually freezes within 10s.

egonl, 2025-04-06:
Can't reproduce it when the *Disarm grace period* is set to 0.
Can't reproduce it on a Nexus.

plesogor, 2025-04-06:
Confirmed was able to reproduce on G474 Matek. Armed during RF Suite LUA change, disarm, wait 5-10 seconds, and lockup occurs every time.

gongtao0607, 2025-04-07:
It seems the newly introduced W25N_NONBLOCKING_WRITE caused the issue. https://github.com/rotorflight/rotorflight-firmware/blob/dc47e428948df48aaf5babd44d1fa305bd4a61ac/src/main/drivers/flash_w25n.c#L40
After disabling it the issue went away.

Check this test build with nonblocking write disabled:
https://github.com/gongtao0607/rotorflight-firmware/actions/runs/14298232368

pmattila, 2025-04-07:
`W25N_NONBLOCKING_WRITE` disabled on G4 for the time being

closed 2025-04-08 by pmattila