rotorflight/rotorflight-firmware #282: Hobbywing V5 telemetry doesn't see current and capacity

alrusov, 2025-04-06:
ESC Hobbywing Platinum 260

[image]

Preliminary note: with version 4.3.0 everything worked as expected.

```
# version
# Rotorflight / STM32F7X2 (S7X2) 4.3.0 May 15 2024 / 09:06:57 (05570fe) MSP API: 12.6
```

I updated to version 4.4.1 and made the appropriate telemetry settings.

```
# version
# Rotorflight / STM32F7X2 (S7X2) 4.4.1 Feb 23 2025 / 15:13:58 (456633b) MSP API: 12.7
```

[image]

And now I don’t see the current and the consumed capacity even in RF Configurator:

[image]

And there is none in the transmitter too

[image]

jimmy6616, 2025-04-09:
The capacity will always show 100% until your start to draw current, the HW will only show current when a load is present, i.e. spoolup with blades on

alrusov, 2025-04-09:
Oh... You're right. Sorry for my inattention.

closed 2025-04-09 by alrusov