rotorflight/rotorflight-firmware #301: Futaba BPS Sensor on RF007 Issue

plesogor, 2025-05-03:
I've got a Futaba BPS rpm sensor installed and connected to Port A for 5v. I've connected the signal pin of the BPS to the TX pin of PortA

I have done this remapping below to remove UART4 from using the pins, and remapped FREQ 1 to A00. 

From reading the FlyDragon posts of users getting this to work. I tried setting both freq_input_pull to PULLDOWN and freq_input_edge to RISING but both parameters do not successfully change, and after saving/reboot they change back to the default PULLUP/FALLING.

These are the commands I've used thus far.
resource SERIAL_TX 4 none
resource SERIAL_RX 4 none
resource FREQ 1 A00
set freq_input_pull = PULLDOWN
set freq_input_edge = RISING

pmattila, 2025-05-04:
I have tested this with 4.5.0-RC3 on FlyDragon and RF007.

**I can't replicate the issue**

After giving the commands, the freq input is configured as instructed:
```
# get freq_input
freq_input_pull = PULLDOWN
Allowed values: NOPULL, PULLUP, PULLDOWN
Default value: PULLUP

freq_input_edge = RISING
Allowed values: FALLING, RISING
Default value: FALLING
```

plesogor, 2025-05-04:
This is a new RF007, I've simply remapped the FREQ 1 sensor to A00. Then tried to execute the commands above. It wouldn't change the settings. I then moved it back from PortA to RPM servo header connected a Spektrum Backplane RPM sensor and it works. Not sure why the CLI would be doing this, i will provide dump/diffs, maybe there is something wierd with my setup.

plesogor, 2025-05-04:
All is working now with Futaba BPS and RF007. Both freq_input_pull and freq_input_edge are working as expected, but I had the FREQ1 mapped to A00, but it was really A01 after discussing with @pmattila .

resource FREQ 1 A01
set freq_input_pull = PULLDOWN
set freq_input_edge = RISING

closed 2025-05-04 by plesogor