rotorflight/rotorflight-firmware #302: Telemetry sensor BEC1  Voltage  - Invalid reading

jimmy6616, 2025-05-05:
Ethos 1.6.3, RF RC-3, 

Incorrect readings of alternate Bec voltages as below, affecting F.bus and CRSF

Ethos  - BEC Voltage - Sensor ID 0901 reading correctly
              BEC1 Voltage - Sensor ID 0219 Invalid reading

CRSF - BEC Voltage - Sensor ID 1081 reading correctly
             ESC1 Bec Voltage - Sensor ID 1049 Invalid reading

In both cases the configurator is showing a correct BEC voltage

[image]
[image]
[image]

rburrow87, 2025-05-05:
Which ESC is this?

jimmy6616, 2025-05-05:
Hobbywing v5

robthomson, 2025-05-05:
@rburrow87 are you going to check it out?

jimmy6616, 2025-05-05:
Its the same with BLHeli32 on an Escape32 ESC

robthomson, 2025-05-05:
It will be the sensor code on fbl not using correct sensor.  @pmattila ...   I can tackle this but am not familiar with code on this front.  Maybe quicker for you?

robthomson, 2025-05-07:
Potential fix here:

https://github.com/rotorflight/rotorflight-firmware/actions/runs/14881872445

robthomson, 2025-05-11:
I have not seen same issue my end

jimmy6616, 2025-05-11:
Rob ive created a new base config on the Nexus, this time with the X20, same result, im using the default RC4 software, no PR's

jimmy6616, 2025-07-19:
I need to check, I dont believe it has, will update tomorrow

closed 2025-11-09 by pmattila