rotorflight/rotorflight-firmware #305: No `mAh` telemetry in FrSky RF007

azaz44, 2025-05-09:
I was just setting up a heli using RF007.
RF 2.1.1, FW 4.4.1.

I could not get "consumed capacity in mAh" telemetry, although it was selected in 'Receiver' tab:

[image]

The workaround is to add a "Custom DIY sensor" which shows as "5250" (if I remember correct), and then need to setup units manually.
I think I had exactly same issue on the other heli when setting it up on Nexus some time ago.

So everything works now, but maybe you want to know about this problem.

-----------------

Also, many telemetry entries (Current, Battery Voltage) are duplicated. Wondering why this happens.

jimmy6616, 2025-05-09:
Your using quite old RF and configurator options ? The reason for multiple sensors is some ESC's dont send telemetry so your seeing both the default Battery sensor for example and ESC battery sensor.

azaz44, 2025-05-09:
> default Battery sensor for example and

What options could be old? Not sure if I follow.

RF is 2.1.1 and FW 4.4.1 so both newest non-beta as of now. This is what I get on telemetry screen - Current and VFAS are duplicated, and both get refreshed when heli is powered on.

[image]

And for consumed capacity I have to make this:

[image]

Which works fine, but might be tricky to find for many users.

Do I have wrong settings anywhere?

jimmy6616, 2025-05-09:
I think you need to publish your diff all in discord under quick help or general. advise what your setup is and let the group reply. Add in your post; TX/RX, F.bus or CRSF, 007 and ESC with firmware version