rotorflight/rotorflight-firmware #309: Total blade pith limit for boom strike prevention

azaz44, 2025-05-17:
This is just an idea.
Since I got Logo 800, I always wonder about some way to make hardware prevent boom strikes, instead of pilot, which has to think about not doing certain types of maneuvers. I think such feature would be useful for big helis like 800, but also for people flying low RPM.

Strategy would be to limit total blade pitch limit for blades when they are above the boom (or 90 deg earlier). Boom strike will happen if there's a lot of negative collective + a lot of down elevator given by the pilot

So similarly to this option:

[image]

One might have another option, like "**Total down-elevator blade pitch**" to set a lower limit, like 20 deg.
This means, while on negative collective, heli would have less cyclic range in down elevator then in up elevator.
Does this make sense?

So far I'm experimenting with achieving desired result by limiting elevator servo range. But I think having this in the mixer would work much better and be very convenient to set.

etocii, 2025-08-06:
I think we need to test whether it's positive or negative blade pitch that causes boomstrike. So far I believe it's actually positive which causes the blade to lag behind.

pmattila, 2025-11-07:
I am no expert... but it appears a boom strike can happen when the disk is tilting backwards (elevator up), especially with lots of negative collective.

How the rotor disk is tilting depends on the dampers. With soft dampers the disk will tilt mostly with elevator. With hard dampers, it tilts more sideways with elevator. Straight tilt backwards would require elevator up and a bit left roll (I think).

But please explain why you think negative / positive pitch is the issue here!

pmattila, 2025-11-10:
I wonder if we are talking about the same thing.
Elevator up == pulling the stick down